Team138Entropy/Entropy2020 #29: Define MODES of Operation

joecaraccio, 2020-01-18:
All hands on deck issue here! We need to think a little bit about how the Robot actually operates during TELEOP, and various modes of what the operator would want to control. I drew up a initial draft for discussion 

`<blockquote class="imgur-embed-pub" lang="en" data-id="a/lnBKI5R" data-context="false" ><a href="//imgur.com/a/lnBKI5R"></a></blockquote><script async src="//s.imgur.com/min/embed.js" charset="utf-8"></script>`

joecaraccio, 2020-01-19:
Correction to that table.. I don't think we will have elevation control, but adjusting exit velocities will be the same thing

Team138Entropy, 2020-01-19:
Joe, do you have this in a document we can all collaborate on? Maybe we could add this to the Wiki as a Markdown table until we have the ConOps document ready.
Comments:
- How do you envision enabling  /disabling auto modes? The controllers will probably be a little cluttered with the manual controls, so I would suggest virtual switches on Shuffleboard
- Should **Manual Acquire** be enabled by default since **Harvest Mode (Auto Acquire)** is disabled by default?
- In seasons past we have assigned operational items like intake (acquisition) to the operator. On your chart you have the entire **Manual Acquire** assigned to the Driver. I suggest splitting this one between the Driver and Operator
- For the **Auto** operations I suggest denoting **Robot** as the _Controller_

joecaraccio, 2020-01-19:
https://drive.google.com/file/d/0B_Tq6mYckE93aEdTbWUweVhGVkoxajRWSzYtTktVWlB4Skc4/view?usp=sharing

joecaraccio, 2020-01-19:
Definetly not virtual switches. Triggers on operator and driver controllers

Team138Entropy, 2020-01-19:
Regarding virtual switches on the Shuffleboard:
- We're going to run out of buttons because they'll likely be assigned to manual operations
- Switching from Auto to Manual or vice versa _shouldn't_ be something that the operator does very often. If they do that inadvertently, the Robot will start behaving in an unexpected manner to the operator, so we'll need to assign buttons that are out of the way.

joecaraccio, 2020-01-19:
Long presses or double presses would solve that. Latency of the driver or operator having to find the laptop mouse and click would be brutal mid match

closed 2020-02-29 by maxh65000