Team138Entropy/Entropy2020 #33: Add toggle to enable/disable LED rings

maxh65000, 2020-01-21:
- [ ] interface with the LED with software
- [ ] keep the second LED ring on unless we see a target within 20 feet

maxh65000, 2020-02-12:
@Nick-P-Orr we're gonna do this with the operator trigger, right?

Nick-P-Orr, 2020-02-12:
Yeah, when the "aim" button (I believe left trigger as of now) is pressed, turn the spike relay on. I don't intend to do individual control of the rings anymore - just always turn both on.

joecaraccio, 2020-02-12:
If you look at the AlternateStateMachine branch,

when auto aim is enabled the LED should come up.
We also plan to be in Auto Aim by default though (and ideally our vision is good enough that we never leave) so we will need to add a check for that state in RobotInit

closed 2020-02-29 by maxh65000