Team138Entropy/Entropy2020 #40: Path following Autonomous

joecaraccio, 2020-01-21:
254 has famously used relatively simple physics concepts to help manipulate their robot at a high level.

One of which is using Motion Profiling to follow paths at a highly efficient level.
We should look into implementing this path following technology into our robot, especially for usage in autonomous. 

I tagged both Liam and Will, but the more the merrier! 

ttps://www.ri.cmu.edu/pub_files/pub1/kelly_alonzo_1994_4/kelly_alonzo_1994_4.pdf

this should be helpful to get an understanding:

https://github.com/Team254/FRC-2019-Public/blob/master/src/main/java/com/team254/lib/control/AdaptivePurePursuitController.java

Barrelwolf38087, 2020-01-21:
I'm already working on this

maxh65000, 2020-01-23:
Duplicate of #9

closed 2020-01-23 by maxh65000