Team138Entropy/Entropy2020 #48: Add driver macros

maxh65000, 2020-01-23:
@joecaraccio suggested about a week ago that we might be interested in adding a set of driver "macros" that allow the driver to press a button and execute a predefined set of movement controls (eg. going around a bot running defense). This sounds like an interesting idea. Any thoughts / macro ideas / volunteers?

joecaraccio, 2020-01-23:
I think we definitely have the man power to pull this off  and if we can do this well would be something fun to do.  Max if you want maybe this would be something good to work on (we are probably a week out from having a turret to continue to integrate on)

joecaraccio, 2020-01-23:
Off the top of my head... I would use the DPad for Macros.... make the direction you hit make sense...

Left Dpad - Left Robot Spin
Right Dpad - Right Robot Spin
Up Dpad - ???
Down DPad - ???

These 'macros' would be good to do when we are in contact with another robot. so I think we would assume our approach would be perhaps heading straight on to a robot

maxh65000, 2020-01-24:
I can't currently work on these macros until #9 is done, because I need stuff like `driveForward(Units.meters(2)))` to be ready. I'll look at this later.

joecaraccio, 2020-01-24:
Place holders.. if you have time just work on it and keep it on a branch for a rainy day

maxh65000, 2020-01-24:
Sounds good

joecaraccio, 2020-01-24:
Would rather see you put your talents to work on this than wasting time on configs/ and CI stuff.

maxh65000, 2020-01-24:
I'll make a dummy subsystem or class to mock this