Team138Entropy/Entropy2020 #52: Refactor Constants.java into real constants

maxh65000, 2020-01-24:


joecaraccio, 2020-01-24:
Why are the constants now marked as deprecated?

maxh65000, 2020-01-24:
to Refactor Constants.java into real constants

joecaraccio, 2020-01-24:
okay.........got that part of it...putting a final tag in front of them makes them impossible to overwrite... what are you thinking?

joecaraccio, 2020-01-24:
Basically I am asking what a real constant is, lol

maxh65000, 2020-01-24:
I am planning to refactor all usages of Constants.java into calls to our Config system but am holding off on that until more people merge their code to avoid creating merge conflicts.

joecaraccio, 2020-01-24:
That would add latency to reads on many values we will probably never change.  A real constant is exactly what we have now. We need your efforts on other much more difficult tasks!

maxh65000, 2020-01-27:
Based on my testing, it takes about ~0.01ms per config read. I think we should be fine.

joecaraccio, 2020-01-27:
So the config in general is a good thing to have. It is good to have a way to change values that we are going to configure quicker.

Here is the thing though... many of the values in the config are not going to be dynamic or really every changed at a point where we need to store them in a file. 

By defining values in final variables we allow the compiler, not only are the reads faster (we know exactly where they are rather than performing a lookup), but we allow the compiler to optimize these values using whatever tricks it has. A simple one if we turn off some feature of a robot, say by setting a boolean false... a compiler could straight up remove blocks of code guarded by that boolean value.

While the latency is kinda splitter hairs here, it looks like based on the number of reads, we could be access that config map more than 10 times alone per loop (the turret subsystem already has 3 calls alone). Eventually that latency could add up to latency we just don't need. 

Because I am loosing my mind I wrote a simple program to test my theory:
`public class App {
    public static void main(String[] args) throws Exception {

        final int TestVal1 = 0;

        Map<String,Integer> map =new HashMap();  
        map.put("key1",1);  
        map.put("key2",1);  
        map.put("key3",1);  
        map.put("key4",1);  
        map.put("key5",1);  
        map.put("key6",1);  

        System.out.println("Read Map 10 Times!");
        for(int i = 0; i < 10; i++){
            long t1 = System.nanoTime();
            int value = map.get("key1");
            long t2 = System.nanoTime();
            System.out.println(t2 - t1);

        }

        System.out.println("Read Variable 10 Times!");
        for(int i = 0; i < 10; i++){
            long t1 = System.nanoTime();
            int value = TestVal1;
            long t2 = System.nanoTime();
            System.out.println(t2 - t1);

        }

    }
}`

with an output of: 

`Read Map 10 Times!
4175
1052
235
198
251
195
200
261
421
320
Read Variable 10 Times!
168
192
164
122
127
141
163
133
127
121`

The first access to the map in particular is at least 20 times slower than a final variable.

joecaraccio, 2020-01-27:
The good news however, is we can still use your forward thinking of using a config file.

As long as final variables are defined in a constructer (so java can guarantee that they are defined), they could be defined at runtime. 

ex:
` public static final double MaxConstant;
    static {
        MaxConstant = 0;
    }`

What if instead, on the RoboRio we store some standard config file somewhere in the same directory as the java application. That way the code always knows at the start of the constants class to attempt to open that file, then for every value we deem the need to have to be dynamically configured, we attempt to parse from the file. If the value does not exist in the file, or we are unable to parse it to the desired type.. prove a fallback value.

I think this solution was talked about, but this would allow us to bypass unneeded latency of data structures.

Barrelwolf38087, 2020-01-27:
That's very close what we have now. Check the structure of the Key enum.

joecaraccio, 2020-01-27:
Except it's not. Upon readbacks you have to perform a bunch of logic and casts to get your value. Check the 'getDouble' function.

Barrelwolf38087, 2020-01-27:
That's a single cast, and it's temporary until we can properly implement static type checking with the default value. The logic isn't even complicated, as it's pretty much just checking if the key exists in the Properties object that's already in memory. I do notice that we're re-parsing the value from the String on every access, and we can improve that by caching the parsed value. Of course, anything robust is going to have *some* level of overhead.

joecaraccio, 2020-01-27:
Okay so tell me how that would be faster than pointing at a value in memory then? 
The solution I proposed would allow you to load in values at runtime with a much lower time penalty. 

No casts would be nesscary on this other solution.

Barrelwolf38087, 2020-01-27:
First of all, we *are* pointing at a value in memory. Everything here is already in memory at this point. Second, what you're proposing is more or less the same thing but split into two separate systems, one for dynamically configured values and one for "true" constants. "If the value does not exist in the file, or we are unable to parse it to the desired type, provide a fallback value" is also an accurate description of what the getter methods already do. If you're just proposing splitting off constants that will never need to be dynamically configured into their own class, then I agree with that.

joecaraccio, 2020-01-27:
You guys are over implementing something that basically already exists in the Java language and are going to experience worse performance doing so. You can believe whatever you want to believe at this point.

Barrelwolf38087, 2020-01-27:
Your idea of worse performance seems to be sensitive down to the nanosecond. In your own example, the *greatest* difference between the HashMap and the static value came out to less than a hundredth of a millisecond. Nitpicking over that is a textbook example of over-optimization. If we *really* cared about *that little execution time*, we would be using C++.

joecaraccio, 2020-01-27:
Max was seeing even more latency from his testing. I won't bother arguing with you guys, you have a lot to learn.

closed 2020-02-15 by maxh65000