Team138Entropy/Entropy2020 #55: Remove the Linter

joecaraccio, 2020-01-24:
Going to have to be the villain here for a second... I don't want to be harsh, but the linter is causing more harm than good. It is removing imports, and doing other unneeded activities... all to make sure spacing is slightly correct.

I think we need to get back to fundamentals here a bit.. instead of focusing on things like linters, configuration control, and other 'git olympic' tasks.. lets focus on the ROBOT.

Here are a few ideas (we may already be working on):
- Motion Control... How can we push this robot movement to its capacity while keeping the it in control
- Way Point Navigation with Action Hooks
- Parallel State Machines... How can we manipulate multiple subsystems at the same time (sorta)
- Control Modes... how are we going to actually use the robot during teleop. Can we make things less manual for the robot?

At the end of the day these are things that will (in my opinion) more exciting to implement.

maxh65000, 2020-01-24:
The linter only removes imports that aren't being used. If you run it locally, you should have no merge conflicts. We are planning on removing the on-push linter and using the git hook local linter.

Barrelwolf38087, 2020-01-24:
The CI linter is only annoying if you don't format it locally, and we have a git hook to do that on every commit automatically. Check the README in master.

jhinspeter, 2020-01-24:
OK. Stop. I'm closing this issue. It's not an issue. We can talk about this all as a team tonight or tomorrow morning.

closed 2020-01-24 by jhinspeter