Team138Entropy/Entropy2020 #56: Azmith Edge Case

joecaraccio, 2020-01-24:
Right now it is looking like the turret will have a +/- 135 rotation from the center point. 
the Edge of this range will be marked by shorting limit switches so we can't rip out wires on accident.

We need to put together some sort of logic in our rotaton class that will be smart enough to know when we reach the edge, but know if we rotate back around the other way we will have the shot.

maxh65000, 2020-01-24:
I'll move this to the project

Nick-P-Orr, 2020-01-24:
Just as a side note, camera FOV can be set to 56º or 75º, but if were going with 75º that means (a little over) ~1/4 of our range of motion will be visible at a time.

Thinking we maybe make it so the turret starts doing the relative positioning when the operator holds down the aim button (ie. Gyro says 190, so we get the turret moving from its position at 135 to 225)? Not sure how our battery draw will be looking if we are also keeping the acquire on most or all of the match, and it is just added wear to the turret. My reasoning is just that if the driver is going to acquire a ball, and the turn in place and the gyro read 190, to 170 to 190 and they do this a few times, the turret would be spinning back and forth a couple times. Not sure if this actually something we need to worry about though.