Team138Entropy/Entropy2020 #57: Define Paths for Path Following

joecaraccio, 2020-01-24:
This year we hope to use a waypoint based autonomous system. Building this out now will allow us to quickly build up many complex operations for all sorts of fun autonomous scenarios. 

Right now I believe Will is working on some sort of interpreter for these paths, but it would be good to have other people familiar, as well as build paths to be used with. 

Some prompts to keep in mind when designing these paths:
- What are less obvious scenarios in autonomous?
- What are some ways we can work well with other high level teammates in autonomous? 

254 shows an excellent example of how quickly we can generate complex paths using a waypoint system.
https://github.com/Team254/FRC-2019-Public/tree/master/src/main/java/com/team254/frc2019/paths

WPILIB also has a 'PathWeaver' which is a java desktop app for generating motion paths:
https://github.com/wpilibsuite/PathWeaver

maxh65000, 2020-01-24:
Does PathWeaver work for 2020? The screenshots were for 2019 but they might be out of date.

joecaraccio, 2020-01-24:
Would be something good to explore. The main WPILIB developer was still supporting it as of 9 days ago so probably. Not super necessary as we can still design them by hand.. might just be nice to visualize though. 

Lol, you love the labels.

Nick-P-Orr, 2020-01-24:
May want to rope Henry in on the autonomous paths. Not sure how many he came up with, but I’m sure he’d appreciate us taking a look at what he has.

PR #151 (2020-03-06): Add auto
Fixes #154 
Fixes #57
closed 2020-03-06 by maxh65000