Team138Entropy/Entropy2020 #62: Shuffleboard VS Smartdashboard

joecaraccio, 2020-01-26:
So from my understanding.. shuffleboard is just a newer, improved smartdashboard? Only reason I ask is because I think some branches are using smart dashboard still. 

Should we convert all of those to shuffle board instead? tagging everyone just to make sure IF they are using the dashboard/shuffleboard they input their feedback

maxh65000, 2020-01-26:
If I understand correctly, shuffleboard is controlled by SmartDashboard API calls.

joecaraccio, 2020-01-27:
https://first.wpi.edu/FRC/roborio/release/docs/java/edu/wpi/first/wpilibj/shuffleboard/Shuffleboard.html

onedeadeye, 2020-01-27:
AFAIK, we only would care about the discrepancy between SmartDashboard and Shiffleboard if we wanted to access a Shuffleboard-exclusive feature from code, e.g. setting the true/false color of a boolean display, in code. We were very inconsistent last year with this and ran into no issues. All beautification is typically done in the Shuffleboard's local configs, which we then save to master.

joecaraccio, 2020-01-28:
Ah okay. Good to know. I'll close this. Thanks Liam

closed 2020-01-28 by joecaraccio