Team138Entropy/Entropy2020 #72: Add toggles for compressor and shifter

maxh65000, 2020-01-31:


joecaraccio, 2020-01-31:
Good call Max. The shifter is going to be plumed 'high' which means in the no air pressure state our robot will be in high gear.

That is good design because that way we can't enter some bad 'low gear only' state.

It would be good to handle a few things when we shift:
- Make sure we have enough air pressure to shift BOTH shifters. I am hoping we have some sort of readback capability. We don't want half our drive base to be in low or high.. in theory both pistons would push out, but if one gear caught and another gear didn't 
- Avoid accidental shifter presses.. my vote would be putting this toggle on the start button and the select button of the driver controller

maxh65000, 2020-02-08:
We're gonna purge this now. #92

closed 2020-02-08 by maxh65000