Team138Entropy/Entropy2020 #96: Intake Conveyor Load System

joecaraccio, 2020-02-10:
https://www.youtube.com/embed/tHYXHcb7rxY?autoplay=1&feature=oembed&wmode=opaque&start=76

^ watch this 10 second video clip. This is EXACTLY what we need to pull off. 

This is exactly what we need to pull off with our load in system. It needs to be as efficient as this system in the video. We likely don't need to bother with the beam sensor.

The flow of this needs to be

- Front Intake Roller goes under load, triggers the storage rollers
- Storage rollers at a certain speed rotates a set number of ticks on the encoder

we can then tune this value so we create a desirable amount of space between balls with allowance for error, while keeping it tight to let us store 5 balls. It will likely be easier to debug this without the state machine and ball counting system. We don't really need to care about the ball count with this method.

jhinspeter, 2020-02-10:
Joe - I agree with you about simplifying if we don't need the beam sensor. I think it can even be as simply as a timer elapsed. Then you don't even need an encoder.

I don't agree with ditching the State Machine - that's intended to **reduce complexity**. Without it the conditional logic will spiral out of control. And I think we do need to count balls, otherwise we'll continue to intake balls when there's no more room.

The idea of the state machine is to automate the intake of a preconfigured number of balls - no operator involvement. I'm willing to pay the cost of a slightly more complex software if it makes life simpler for the operator.

maxh65000, 2020-02-10:
Are there currently encoders on the intake

joecaraccio, 2020-02-10:
> I don't agree with ditching the State Machine - that's intended to **reduce complexity**. Without it the conditional logic will spiral out of control. And I think we do need to count balls, otherwise we'll continue to intake balls when there's no more room.

I said for debugging to make things simpler.

joecaraccio, 2020-02-10:
> Are there currently encoders on the intake

Not yet. Nick L talked about adding one though so I think we will get one on tonight

joecaraccio, 2020-02-10:
Probably would still need an encoder. Its pretty simple for them to add. With different loads of balls it might take longer to drive if you were using strictly a timer. Encoder would be a more concrete solution that would probably be just as easy to implement as dealing with a timer.

closed 2020-02-22 by maxh65000