Team138Entropy/Entropy2020 #134: Climber System Details

joecaraccio, 2020-02-27:
The Mechanical Team has basically complete the climber. But with that comes concerns for when we use the climber:

- We only have one set of arms. We can't bend this thing or we are dead in the water. There is a concern if we have the arms up all the way, the driver could bend them backwards on the bar driving too fast into it.**When we are extending the climbing arms, we should slow down the drives throttle ability to prevent this.**

- There were talks of needing a climbing camera. Can we come up with a solution that only turns on the camera when we need it, so we don't have to sacrifice streaming quality of the other camera

- Climb Controls interlock. We should make the operator have to hit 'select' or 'start' or something on their controller before they are able to access the climbing controls. We don't want any accidental usage of the climber.

maxh65000, 2020-02-27:
1. That sounds like a good idea.

2. We can support up to 2 cameras on the robot (and possibly more) and these camera instances can be created or destroyed at any time. We could have the frontCamera be replaced whenever you toggle to climbing mode from Camera 0 to Camera 1. See https://github.com/Team138Entropy/Entropy2020/blob/master/src/main/java/frc/robot/subsystems/CameraManager.java#L40 for an example of how these camera instances work.

3. Don't we have something that switches us into a climbing mode? Is this sufficent?

joecaraccio, 2020-02-27:
1. That could just be a simple parameter applied to the drive loop.

2.  I think that is how we would do it. Use a camera we would otherwise never need. I don't want to ruin the bandwidth of the drive camera. Just when we are in the endgame.

3. Don't have it yet, it could just be a boolean in Robot that has to get toggled.  I just think we don't want to accidentally send the climber up.  I also need to verify that the turret can be in any configuration when climbing. I know initially there were concerns that we would crush one of the motors on the shooter, so we may need to send the turret to a certain orientation before climb.

maxh65000, 2020-03-01:
These are in their own issues now.

closed 2020-03-01 by maxh65000