Team138Entropy/Entropy2020 #156: Climbing Changes

joecaraccio, 2020-03-02:
Because we don't have limit switches, we can't really use any sort of homing function with the climber unfortunately.  We are going to have to use test mode to manually move the climber down to its retracted state in the pits prior to a match.

On Robot boot, that can then become the zero location of the encoder. From measuring yesterday, we know we should not go above 75,000 encoder ticks.  I simplified the State machine a little bit so we only have one constant loop:

**- A Start interlock button allows the Driver to get into climbing mode. This allows the Extend and Retract climber buttons to actually be used**
**- Check if the Operator is commanding the climber to rise, if its below the maximum ticks allow it to do so
- Check if the Operator is command the climber to lower, if it is above 0, allow the operator to do so.** 
**- Else if the operator is not commanding to do the climber to do anything, hold it in that location and wait for the command**

Instead of using Position mode, we should use a slow percent output as we can control the speed when doing so. In between Competition 1 and Competition 2 we can explore positional control.

We can then deal with manually retracting it and getting it reset in test mode.

jhinspeter, 2020-03-02:
We already have test operation for the climber mapped like the other motors. It requires the operator be in test mode, hold down (one of the bumpers I think?) and then use the left joystick axis to provide manual percent output based on the joystick position.

joecaraccio, 2020-03-02:
Ah okay awesome that we already have that. From using it though I would have the percent output be very low. We overdrove the climber this weekend and it resulted in over an hour long fix for mechanical. 

I would say just zero it at the start of the match based on its current position. We can set the position in test mode prior.

closed 2020-03-05 by maxh65000