Team138Entropy/Entropy2020 #161: Allow climbing escape?

joecaraccio, 2020-03-04:
Right now we have an interlock on the climbing mode, which requires to the driver to enable climbing before then being able to be able to use the climbing controls

- do we think it would be a good idea if the back/select button is pressed to leave climbing mode?

Or perhaps a long press to enter climbing mode?

I have fear of this doomsday scenario where early in a match we actually put ourselves into climbing mode and can't get out of it.

I guess if we start running the climber than none of this would apply

onedeadeye, 2020-03-04:
Perhaps this didn't get merged or my untested code from Saturday doesn't work (surprise!) but i built in a mechanism to leave climbing mode - if the climber is all the way down, pressing the "climbing mode" key transfers back out to Intake.

jhinspeter, 2020-03-04:
Liam - yes, you're right. That was in there but I took it out while I was
trying to get climbing working. Switch statements need breaks. ;^)

I think we could probably put it back in, now that I have climbing working.
Or as Joe says, make it more difficult to enter. We should get input from
Noah.

A clear indication of the current active state on the dashboard might help
too. Like a green light next to the active sub-state. Alternatively, can we
make the main state display bigger text?

joecaraccio, 2020-03-04:
I would just say for simplicity (we are close to the deadline) to allow the select button be hit again to allow you to leave the 'Wait' state.  if you are beyond that state then you have to climb.

Jay - DisabledInit is probably our best bet for calling a zero function. that should run everytime we turn on I believe.. and anytime we disable..

PR #162 (2020-03-05): Integrate climber
Fixes #161, #156, #152
closed 2020-03-05 by maxh65000