Team138Entropy/Entropy2020 #183: Revise Boat Mode

joecaraccio, 2020-03-08:
We never got the chance to before the competition, but the center of gravity is MUCH better on the competition robot, thanks to the Winch, and battery placement. The drivetrain also uses mini cims, which should have less power than regular ones.

- Do we still need boat mode?

- Can the boat mode values be changed to allow more acceleration?

joecaraccio, 2020-03-12:
Assigning this to Liam -

Liam.. Boat mode worked awesome on the practice robot and is still needed, but I think we might not want it on the competition robot. On an experimental branch, could you revert everything related to drive back to 254's cheesy drive algorithm, try that... have the drivers try it and see if there are any differences? I suspect based on the CG improvements and loss of power with mini cims it won't be as easy to tip.

onedeadeye, 2020-03-12:
Can do!